jcyrax/pcbmodelgen #1: Can't use example script

KhazAkar, 2019-03-19:
OS: OpenSUSE Tumbleweed
Octave 5.1.0
openEMS 0.0.35
`openEMS FDTD startup
Using Octave script files
error: 'physical_constants' undefined near line 24 column 1
error: called from
    simulation_script at line 24 column 1
octave:1>`

jcyrax, 2019-03-21:
You need to add OpenEMS matlab share to your octave path. I have made file .octaverc in my home directory
```
addpath('/home/janis/ProjectsContrib/openEMS/out/share/openEMS/matlab')
addpath('/home/janis/ProjectsContrib/openEMS/out/share/CSXCAD/matlab')
```
physical_constants is defined in OpenEMS, it's used in octave scripts so that's why the complaint.

KhazAkar, 2019-03-21:
Ok, now worked, but It is normal to close AppCSXCAD to perform simulations by OpenEMS?

jcyrax, 2019-03-21:
Yes, thats how its done in examples, so you could verify that geometry is ok, before waiting long time to complete simulation. If you don't need that, just edit octave script file. There is CSXGeomPlot call somewhere inside.

closed 2019-03-30 by jcyrax

mfaccin, 2019-11-20:
Warning: No primitives found in property: box_material!
Warning: No primitives found in property: drill_fill!

@jcyrax, I got these two warnings trying to run the examples. How to solve/add these primitives?

jcyrax, 2019-11-30:
@mfaccin, next time open new issue instead!
Regarding your question, this is just a warning OpenEMS gives to inform that there are no conductors in specified materials. In examples this is no problem. For your own simulations, this can be helpful hint to know something may be wrong with geometry.