jcyrax/pcbmodelgen #2: Accept pr with clang-format?

matthuszagh, 2019-04-14:
Thanks for writing this program!

I'm making some modifications to support kicad 5. I personally use clang-format and was wondering if you'd be open to accepting a pr with formatting changes from a clang-format script. I'll try to make it as similar to the existing style as possible (e.g. 4 spaces indent, etc.). Additionally, are you interested in the kicad 5 pr's when I get those working?

jcyrax, 2019-04-15:
> if you'd be open to accepting a pr with formatting changes from a clang-format script

Do you mean reformat all code base to be consistent using some clang-format style? I have no objections to that, if its about the same as current style, just add the style file in PR.

> Additionally, are you interested in the kicad 5 pr's when I get those working?

Yes, that would be very much appreciated. I currently have no plans and time to develop this any further.

matthuszagh, 2019-04-16:
Here's the [pr](https://github.com/jcyrax/pcbmodelgen/pull/3). Let me know if you'd like anything adjusted.

closed 2019-04-16 by matthuszagh