jcyrax/pcbmodelgen #5: Error in parser GetZone parser

huegene, 2020-04-21:
When I add a keepout zone, I can set it to Top and Botom layer at the same time. 
In that case the line for that zone becomes;
"(zone (net 0) (net_name "") (layers F&B.Cu) (tstamp 0)"
but the parser is looking for the string "layer" and misses the "layer–s–" string and throws an exception 
  what():  GetZone: no 'layer' field 

 preliminary workaround make two seperate keepout zones, one for each layer. 
but a more permanent solution (in the parser) would be much better :-)

thanks for your work
kind regards
huegene

jcyrax, 2020-04-29:
Thanks for reporting. Currently I'm very busy, but will try to look into this problem when I have time.

leoheck, 2020-06-27:
@huegene 
I am having issues with zones and I was trying to fix my issue then I saw your issue, and I think I was able to fix it. 
I am adding this to the PR that I have opened now.
https://github.com/jcyrax/pcbmodelgen/pulls

I am testing this fix and it looks like it is working...
[image]

[image]

closed 2021-09-09 by jcyrax