jcyrax/pcbmodelgen #6: Running it on Ubuntu 20.04

leoheck, 2020-06-16:
Hello, I am testing your tool with Ubuntu 20.04.

I managed to install it and all required dependencies.
Maybe I can document this if you are interested in that.

I also had to add 2 paths on the `.octaverc` so the tools could find the `OpenEMS` and `CSXCAD` files as you said in another issue.

Right now I am running this for the first time using the `test_pcb`

[image]

I hope to be able to run this for my board.

leoheck, 2020-06-16:
Also, I saw that this example was build using Kicad nightly build. It would be nice to port it to the `5.1.*` stable so more people could open it easily. But the fix is straightforward.

I just had to remove these red lines.
[image]

jcyrax, 2020-06-17:
Yes, I agree. Need to update examples to new version.

closed 2021-09-09 by jcyrax