jcyrax/pcbmodelgen #7: Issues with the commands in generate_model.sh

leoheck, 2020-06-16:
Hi, I am playing with this tool for the first time. Now I am running the `generate_model.sh` line by line.

I am seeing these 2 issues. 

```
Warning: kicad_pcb file version not 4. May experience some errorsERROR: mesh gap size too large. Configured max size 1.000000, but actual size 1.322618
```

```
ERROR: mesh gap size too large. Configured max size 1.000000, but actual size 1.200192
```

What I can do fix the error? I don't know where this config is placed. Is this something related to the config.json or the board itself?

[image]

leoheck, 2020-06-16:
Actually I am seeing 2 errors.
I improved the warning output adding a \n at the end.

These are the errors, that I still have no idea on how to fix.
```
ERROR: mesh gap size too large. Configured max size 1.000000, but actual size 1.322618
ERROR: mesh gap size too large. Configured max size 1.000000, but actual size 1.200192
```

leoheck, 2020-06-16:
Ok, I fixed these warnings by changing the config.json

I changed these items from 1 to 1.6 to try. Now I have to understand this config.json.
Do you have any documentation explaining it?

[image]

Also, I am seeing another issue. 
The `pcbmodelgem` is finishing with an error.
This is not good, I will also try to fix it too.

[image]

leoheck, 2020-06-16:
Ok, that was easy, the main is returning a hardcoded 1.

```
    return 1;
}
```

If it is hardcoded it is better to be 0. Otherwise, the OS will understand the execution was failing.

https://www.tutorialspoint.com/batch_script/batch_script_return_code.htm

jcyrax, 2020-06-17:
Hi

Regarding the errors. I'll comment on JSON config. For other problem you're right and I saw your pull request that addresses this issue. 

pcbmodelgen uses config.json to control some aspects of conversion, because some of the information isn't found in kicad pcb or at the time of writing wasn't or I didn't implement. 

There are few parts to conversion:

1. Converting metal geometry to octave script so it can be parsed by openems.

1. Generating mesh lines based on pcb metal geometry also as octave script.

1. Some information in simulation script. For instance material properties, feed points.

Description of some fields in config.json

**pcb_metal_zero_thick** - If this is true, then top and bottom copper has zero thickness. This allows for less mesh lines and faster simulation, but you must be careful how mesh lines are positioned otherwise you can get invalid results.
 
**corner_approximation** - impacts number of mesh lines and so the simulation time.

**insert_automatic_mesh** - this controls automatic mesh line generation. You don't have to use it, you can generate the lines as needed manually or by some other automation process.

**manual_mesh** - insert your manual mesh line positions here. They will be inserted before automatic line generation.

**min_cell_size** and **max_cell_size** - sets needed cell size boundaries for simulation. This relates to your test signal bandwidth. Make as large as possible for used test signal frequency to minimize mesh line count.

**pcb_z_lines** - will insert this amount of mesh lines between top and bottom copper layers, before performing automatic mesh line generation. This is needed because there is no geometry on the inside of PCB.

**smooth_mesh_lines** and **smth_neighbor_size_diff** - this tries to make neighboring mesh cells not differ by more than smth_neighbor_size_diff times. This is needed for FTDT simulation. You can't have abrupt change in cell sizes - this can make your simulation invalid.

**SimulationBox** - sets simulation domain region. You need to place this around your geometry with sufficient distance.

**use_box_fill** - fill simulation domain with specified material.

One thing to notice regarding mesh line generation. Not always it is successful given needed parameters. If you request large min size and small difference between neighboring cells it will fail to comply and report errors you mentioned. This is also highly dependent on pcb geometry. As each geometry point contributes to mesh lines. This in itself doesn't mean that you can't use the results, but you need to evaluate resulting mesh and decide for yourself if given mesh is good enough for your usage case. You can explore generated kicad_pcb_mesh.m to see all warnings about non conforming mesh sizes.

Also if you are using pcb_metal_zero_thick you need to adjust mesh lines accordingly. For that there is boundary_one_third_rule. I'll add more info regarding that tomorrow. http://openems.de/index.php/FDTD_Mesh.html

leoheck, 2020-06-17:
Cool, I am working to improve my PR. I can add this info on my README, so it will be in that PR too.
It is done, but I am testing this and I am just having some issues with my `AppCSXCAD` that I didn't see yesterday.

```
openEMS FDTD startup
Using Octave script files
Model import and simulation setup done
Generating simulation configuration file
Done
Showing geometry
invoking AppCSXCAD, exit to continue script...
malloc(): corrupted top size
../../build/AppCSXCAD.sh: line 5: 99135 Aborted                 (core dumped) AppCSXCAD $*
Starting openEMS
args = "simulation.xml" --debug-PEC --debug-material
/usr/bin/openEMS: symbol lookup error: /usr/local/lib/libCSXCAD.so.0: undefined symbol: _ZN3MPI8Datatype4FreeEv
error: load: unable to find file ./tmp/port_ut1
error: called from
    ReadUI at line 64 column 9
    simulation_script at line 119 column 3
octave:1> coredumpctl list
octave:1> exit
```

closed 2020-06-26 by leoheck