jcyrax/pcbmodelgen #10: How to fix examples from version 5.99 to the current version 5.1.6

leoheck, 2020-06-26:
The `kicad_pcb` created with Kicad-5.1.6 and the Kicad-5.99 (used on the examples) differs in the definitions.

Kicad-5.99, set layer names with quotation marks while the current version 5.1.6 doen't.

To validate this, I opened the `text_pcb` example on my Kicad. Then saved the board, then after diffing it, I saw these changes.

I was able to fix the board (by adding these quotation marks on the layer names) as needed for `pcbmodelgen` since it was fixed using Kicad-5.99 as the reference.

```
#!/bin/bash

board=$1

# Kicad-5.1.6
# - Layers do not have quotation marks
# - This fix "update" the board after saving it with Kicad-5.1.6

sed -i 's/F.Cu/"F.Cu"/g' $board
sed -i 's/B.Cu/"B.Cu"/g' $board
sed -i 's/F.Adhes/"F.Adhes"/g' $board
sed -i 's/F.Paste/"F.Paste"/g' $board
sed -i 's/F.SilkS/"F.SilkS"/g' $board
sed -i 's/B.Mask/"B.Mask"/g' $board
sed -i 's/F.Mask/"F.Mask"/g' $board
sed -i 's/Dwgs.User/"Dwgs.User"/g' $board
sed -i 's/Cmts.User/"Cmts.User"/g' $board
sed -i 's/Eco1.User/"Eco1.User"/g' $board
sed -i 's/Eco2.User/"Eco2.User"/g' $board
sed -i 's/Edge.Cuts/"Edge.Cuts"/g' $board
sed -i 's/Margin/"s/Margin"/g' $board
sed -i 's/B.CrtYd/"B.CrtYd"/g' $board
sed -i 's/F.CrtYd/"F.CrtYd"/g' $board
sed -i 's/F.Fab/"F.Fab"/g' $board
sed -i 's/Margin/"Margin"/g' board.kicad_pcb

# Extra (maybe not needed but it is also different between versions)
sed -i 's/A4/"A4"/g' board.kicad_pcb
sed -i 's/net_class Default/net_class "Default"/g' board.kicad_pcb
```

leoheck, 2020-06-26:
That being said, it is nice to make these changes in the pcbmodelgen parser, maybe accepting both if possible. I don't know how and where to change, but if I have an idea I will do it.

leoheck, 2020-06-26:
Oh! That will be easy, the last change was doing exactly that.

```
    std::string ll;
    if (kicad_version == "4.x")
        ll = "Edge.Cuts";
    else
        ll = "\"Edge.Cuts\"";
```

So, it is just a matter of adjusting the version properly.

leoheck, 2020-06-26:
Alright, fixed. It is in my PR.

leoheck, 2020-06-26:
https://github.com/jcyrax/pcbmodelgen/pulls

iampadik, 2021-09-09:
> Oh! That will be easy, the last change was doing exactly that.
> 
> ```
>     std::string ll;
>     if (kicad_version == "4.x")
>         ll = "Edge.Cuts";
>     else
>         ll = "\"Edge.Cuts\"";
> ```
> 
> So, it is just a matter of adjusting the version properly.

Hi, Can you please tell me where do I make these changes? Where do I find pcbmodelgen parser? 
Thanks.

leoheck, 2021-09-13:
Oh, sorry @iampadik, did not see this before. 
You can change this on this file `source/ems.cpp`, line 864.

The original file here was already fixed I think.  [Source](https://github.com/jcyrax/pcbmodelgen/search?q=ll+%3D+%22Edge.Cuts%22%3B)

[image]

iampadik, 2021-09-13:
Hi Leoheck.

Thank you for your response. Yes, now I crosschecked, the changes are already made. But I couldn't use my kicad 5.1 layouts for simulation. I ultimately installed version 4 in windows, make the whole layout and then use pcbmodelgen, then it works.

May be I should again give it a try and think if I missed anything. Thanks anyway again!!

leoheck, 2023-06-26:
This is a quite old report and since we alredy have Kicad 7,  I am closing this issue.

closed 2023-06-26 by leoheck