jcyrax/pcbmodelgen #11: error: the type for the property "hole_fill" cannot be found

leoheck, 2020-06-27:
I am not sure what is this `hole_fill` do you have any idea what I have to look to fix/understand this property?
I just downloaded one board, that was built with Kicad 4.0 from Kicad demos.
And I am trying to run the simulation using configs (`pcbmodelgen.json`, `simulation_script.m`) from the `test_pcb` example.

> Just a little bit of info about my background. I have zero experience with openEMS. I am just planning to play with it to learn something new. And I found that the pcbmodelgen would be a nice starting point.

```
Check generated mesh file for detailed warning information
octave --silent --persist simulation_script.m
openEMS FDTD startup
Using Octave script files
error: the type for the property "hole_fill" cannot be found
error: called from
    Add2Property at line 12 column 5
    AddLinPoly at line 41 column 5
    kicad_pcb_model at line 5953 column 5
    simulation_script at line 64 column 5
octave:1> 
```

jcyrax, 2020-06-29:
`hole_fill` is material property name used for drill holes and via holes. I think you needed to define it in simulation_script.m. You can try to add:
```
CSX = AddMaterial(CSX, 'hole_fill');
CSX = SetMaterialProperty( CSX, 'hole_fill', 'Epsilon', 4.2, 'Mue', 1, 'Kappa', 0, 'Sigma', 0, 'Density', 1 );
```
There are few materials names that are hard coded and used when generating octave scripts for geometry.
Now I don't exactly remember why I don't have hole_fill in examples, but I'll check it during next few days.

leoheck, 2020-07-01:
> but I'll check it during next few days.

Cool, thanks. Your answer was good enough to understand the context.

jcyrax, 2020-07-03:
Hole fill isn't in examples because there were no vias or drills. If you have vias or drills in your PCB just add definition for that material in your simulation script before line that imports kicad pcb:
`CSX = kicad_pcb_model(CSX);`

You can see that generated file kicad_pcb_model.m has lines like `CSX = AddPolygon(CSX, 'metal_bot', 1, 2, -0.000000, p);` 

There are hard coded material names: box, pcb, metal_top, metal_bot, hole_fill. 
Besides obvious ones `box` is for whole simulation space. This is used in case `use_box_fill` is set in pcbmodelgen.json

closed 2020-07-03 by jcyrax