jcyrax/pcbmodelgen #12: KiCAD to OpenEMS (PCB export)

gnumohan, 2020-12-29:
Hi Janis,

I have tried to convert .kicad to .json and see the below error.

$ pcbmodelgen -p Meander_2_4G.kicad_pcb -c Meander_2_4G.json
Error from reader: * Line 1, Column 1
Syntax error: value, object or array expected.
terminate called after throwing an instance of 'Json::RuntimeError'
what(): * Line 1, Column 1
Syntax error: value, object or array expected.

Aborted (core dumped)
$

Tools versions:
Kicad - 5.1.9
Ubuntu - 20.04.1 LTS

Thanks,
Mohan

jcyrax, 2020-12-31:
Hi

1. Could you please provide Meander_2_4G.json if it is not confidential?
2. Could you test if examples work for you?

gnumohan, 2021-01-02:
Hi Janis,

Sorry for the late reply as i haven't worked for the last 2 days.

>> 1. Could you please provide Meander_2_4G.json if it is not confidential?

Just for checking, I have attempted to take your example pcbmodel.json as it is and simulation_script.m, replaced only the pcb files running using makefile without changing any params and "make run".

Now, kicad_pcb_mode.m and kicad_pcb_mesh.m is generated with mesh gap / ration ERROR messages as attached in the error_log. Attaching here the complete file in .zip.

The Meander_2_4G.json looks to be corrupted as "cat Meander_2_4G.json" showed some invalid ASCII characters at the start of the file.

At this point I am not doing any confidential work and with this attempt would like to publish as part of your example folder/openems as reference for people interested in meanderf_24_g simulation.

>> 2. Could you test if examples work for you?

Yes the examples are working fine in my environment.
Also,
As part of the kicad_pcb_mode.m generated now using your example .json and makefile, 'hole_fill' property is generated but this property doesn't exist now.

>> simulation_script

openEMS FDTD startup
Using Octave script files
error: the type for the property "hole_fill" cannot be found
error: called from
    Add2Property at line 12 column 5
    AddLinPoly at line 41 column 5
    kicad_pcb_model at line 23 column 5
    simulation_script at line 68 column 5
>>

Thanks,
Mohan

[meander_24g_1.tar.gz](https://github.com/jcyrax/pcbmodelgen/files/5760690/meander_24g_1.tar.gz)
[error_log.txt](https://github.com/jcyrax/pcbmodelgen/files/5760691/error_log.txt)

M577APC, 2021-09-14:
Hi,

I think I've found the solution for this problem. In both sample projects **"hole_fill"** is set up in the .json file. Also in both projects no **"hole_fill"** is required for simulation. Therefore both work perfectly. However, in both example simulation_script.m, this material is defined as **'drill_fill'** instead. 
It must be credited, that there is a warning message as well. Still I think it would be better to correct this detail for avoiding further confusion.

gnumohan, 2021-09-17:
Hi,

Ok. Thanks. 

Will try this solution and get back.

montanaviking, 2023-04-21:
My bad. Sorry, a search of the examples   e.g. pcbmodelgen-master/examples/branch_line_coupler folder showed it contains the
pcbmodelgen.json file I apparently needed. Apparently, I still needed to add a mesh. Perhaps I need to edit the .json file for my application too. Looking in how to do that.
Thanks, Phil

montanaviking, 2023-04-21:
Also, I noticed that only the metal polygons in the *.pcb kicad pcb file are not translated, but only the objects drawn using "Route single track". Is there a way to translate all the metal polygons to the *.m file?
Thanks,
Phil

jcyrax, 2023-04-22:
@montanaviking  you need to create configuration json file for your project (you can use example files as starting point). Also normally I would expect there to be all metal geometry.

closed 2023-07-13 by jcyrax