jcyrax/pcbmodelgen #15: Failing to detect port

iampadik, 2021-09-07:
Hi,
I am new to Pcbmodelgen and trying to figure out how to run kicad layout rf simulation in pcbmodelgen. 

I followed the following instructions properly:

_sudo apt-get install libtinyxml2-dev libtclap-dev
mkdir build
cd build
cmake ../
make
sudo make install_

and then, put 

_addpath('/usr/share/octave/packages/openems-0.0.35/')
addpath('/usr/local/share/CSXCAD/matlab/')_

in the octaverc file.

After I try to run pcbmodelgen -p pcb.kicad_pcb -c pcbmodelgen.json for the branchline_coupler_example, I can see the 3d view of it and after i close the image, following error occurs:

_Segmentation fault (core dumped)
error: load: unable to find file ./tmp/port_ut1
error: called from
    ReadUI at line 64 column 9
    simulation_script at line 150 column 3_

Could you please guide me as to why am I getting this error. Also, I don't see any ports defined in the example. How would pcbmodelgen find out where the ports are? The error also asks about a port file in the temp folder.

Can you please clear my doubts here? Thanks.

jcyrax, 2021-09-08:
Hi!

Currently the ports are defined in octave script file (simulation_script.m) with function AddLumpedPort. This means you need to manually set them based on your design. Regarding your segmentation fault I will check if its working for me, but one thing you can do is test if OpenEMS examples are working. Also you can investigate core dump and check backtrace where the fault is occuring.

iampadik, 2021-09-09:
Hi.

Thank you for your reply. I basically reinstalled everything since I was a little leery about my installation process. Now, I can run your examples properly. 

But, I am not able to run it for my new files (both, nightly and Kicad 5.1.6). 

For nightly, metals, PCB,drill, nothing was detected. But the code took up the ports as I defined it in the octave script file. 

For Kicad 5.1.6, it gives me the following error:

_terminate called after throwing an instance of 'std::length_error'
  what():  vector::reserve
Aborted (core dumped)_

Also, how much part of pcbmodelgen.json, simulation_script.m has to be manually edited? I am a little confused what I am missing. Thank you so much in advance.

jcyrax, 2021-09-09:
> For Kicad 5.1.6, it gives me the following error:
> 
> terminate called after throwing an instance of 'std::length_error'
> what(): vector::reserve
> Aborted (core dumped)

You probably need to edit makefile with correct file name of your kicad PCB.

> Also, how much part of pcbmodelgen.json, simulation_script.m has to be manually edited? I am a little confused what I am missing. Thank you so much in advance.

Currently there is very little automation. This tool will convert PCB to octave script, compatible with the way you would normally interact with OpenEMS and also you can get mesh lines. Other then that you will have to do things manually, like setting correct dimensions and port locations and writing script to process measurements and so on.

I have some ideas about further automation, but currently no free time.

iampadik, 2021-09-10:
> You probably need to edit makefile with correct file name of your kicad PCB.

>>I have already done that really.

> Currently there is very little automation. This tool will convert PCB to octave script, compatible with the way you would normally interact with OpenEMS and also you can get mesh lines. Other then [...]

>>Thanks. I will look into it.

> I have some ideas about further automation, but currently no free time.

>>Sure, I understand. No worries. Will wait for it! :)

> Edit: Also check examples and maybe you can get away with extending them.

>>They are working. But I guess, everything in the simulation_script.m of examples are ready.  In my case, the simulation_script.m doesn't get generated by itself after "make"  command.

iampadik, 2021-09-10:
Hi.
Update: 

I copied the simulation_script.m file from example and edited it. 
1. Assigned ports in it, edited the graphs to be plotted. 
2. Changed the kicad.pcb file in makefile.
3. adjusted the mesh settings in the pcbmodegen.json file since it was giving an error of mesh gap

Now, when I run "make run", it gives me the following error after the FDTD engine starts running:

_Engine_Interface_FDTD::CalcVoltageIntegral: Error, only a 1D/line integration is allowed_

I think I am almost there.

iampadik, 2021-09-11:
Update:

So, I thought of posting of how I have for now solved my problem.
I simply 
1. installed everything again properly making sure, octave, openEMS and pcbmodelgen are configured properly for each other. 
2. Installed Kicad-4.0.7 in windows and made my layout there completely.
3. Copied a. makefile, b. pcbmodelgen.json and c.simulation_script.m from a suitable example of pcbmodelgen and copied it in my kicad project folder
4. Made some changes in a. makefile - changing kicad.pcb filename, b. edit the PCB substrate thickness and mesh parameters  in the pcbmodelgen.json file, c. gave the correct coordinates of the ports in the simulation_script.m file
5. and simply run it. 

It works !! Thank you Janis for your support!!

closed 2022-08-10 by jcyrax