jcyrax/pcbmodelgen #16: Problem with bottom pads

M577APC, 2021-09-08:
Tried to convert a 2-Layer pcb with 1.6mm thickness and set 
"pcb_metal_thickness" : 0.1,
"pcb_metal_zero_thick" : false,
Result: All the pads of the bottom layer are floating below the PCB. While the bottom layer itself is on Position 0 in Z-direction, the position of the Pads is exactly -1.7mm.

jcyrax, 2021-09-08:
Hi! If you can share kicad pcb and json config file it would be helpful. Maybe you can isolate or recreate exactly the problem and I will look into it.

M577APC, 2021-09-08:
[pcbnjson.zip](https://github.com/jcyrax/pcbmodelgen/files/7131350/pcbnjson.zip)

Hi Janis, many thanks for your immediate response! Could you also try to implement the recognition of passive components and conversion to lumped elements?

M577APC, 2021-09-09:
[Result.pdf](https://github.com/jcyrax/pcbmodelgen/files/7134583/Result.pdf)

jcyrax, 2021-09-09:
Should be fixed in 3d6ef89af8ef34814bbe6b6624c77854df4b1458

M577APC, 2021-09-10:
👍🙋‍♂️

closed 2021-09-10 by M577APC