jcyrax/pcbmodelgen #20: Bad copper infills (kicad 7.0.0)

ThomHPL, 2023-05-14:
When using a pcb file from kicad v7.0.0, some copper infills are not generated correctly due to some points being stored as "NaN" in the files generated by pcbmodelgen.

I tracked down the issue to a float issue in the function Zone::getVectAngle. I have made a quick fix and will issue a pull request soon.

Full kicad version info:
Application: KiCad PCB Editor x64 on x64
Version: (7.0.0), release build
Libraries:
    wxWidgets 3.2.1
    FreeType 2.12.1
    HarfBuzz 5.0.1
    FontConfig 2.14.1
    libcurl/7.83.1-DEV Schannel zlib/1.2.13
Platform: Windows 11 (build 22621), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    Date: Feb 12 2023 01:35:19
    wxWidgets: 3.2.1 (wchar_t,wx containers)
    Boost: 1.80.0
    OCC: 7.6.2
    Curl: 7.83.1-DEV
    ngspice: 39
    Compiler: Visual C++ 1934 without C++ ABI
Build settings:
    KICAD_SPICE=ON

leoheck, 2023-06-26:
Hi guys, is this issue resolved?

ThomHPL, 2023-06-30:
Hi, my fix was merged and it works flawlessly for me. Let me know if you encounter issues.

leoheck, 2023-07-01:
Sure, this issue can be closed then. If I have issues I can open a new issue.

closed 2023-07-01 by ThomHPL